s-matyukevich/raspberry-pi-os #40: Use kernel8.img naming convention

evopen, 2018-06-18:
Pi 3 supports the kernel8.img naming convention. [Here](https://www.raspberrypi.org/documentation/configuration/config-txt/boot.md)

I think by using kernel8.img, we can drop the arm_control flag (I drop it, turns out ok). 
After all we have ARMv8.

s-matyukevich, 2018-06-18:
This makes sense, I'll update the doc and Makefile

evopen, 2018-06-23:
@s-matyukevich I can make pr if you like. But I need confirmation that this naming thing works with raspberry pi 3. Because i own a 3 b+.

s-matyukevich, 2018-06-23:
@evopen a PR would be perfect! I already tested kernel.8 on my raspberry pi 3 b and it works fine.

In case you proceed with a PR please don't forget to update the Makefiles for all lessons and in the text of lesson 1 mention arm_controll flag as well - otherwise people might be confused when they see exercises based on old arm_control approach.

s-matyukevich, 2018-06-24:
Done https://github.com/s-matyukevich/raspberry-pi-os/pull/59

closed 2018-06-24 by s-matyukevich