s-matyukevich/raspberry-pi-os #56: lesson 1: mismatch in uart_init()

Tommac1, 2018-06-22:
Hello,

in the code snippet for **Mini UART initialization** we have line setting register _AUX_MU_IIR_REG_:

`put32(AUX_MU_IIR_REG,6);                //Clear FIFO`

Although, it is not described, what it does and it is not present in the source code for lesson 1.
Should it persist in the code (and what it does?) or should it be dropped?

Thank you in advance.

s-matyukevich, 2018-06-22:
Bits [1:2] of the registers can be used to clear UART internal character queue. After UART is enabled the queue is already empty, so this line is redundant. I'll delete it to avoid confusion.

s-matyukevich, 2018-06-22:
Fixed https://github.com/s-matyukevich/raspberry-pi-os/commit/8f58dbd3c9e43c7898557355977aab77944918fe

closed 2018-06-22 by s-matyukevich