s-matyukevich/raspberry-pi-os #65: [Lesson01] Base address computation

T4g1, 2018-06-25:
I'm trying to figure out the computation behind the given base address (in https://github.com/s-matyukevich/raspberry-pi-os/blob/master/src/lesson01/include/peripherals/base.h). Can someone explain how to go from "The Raspberry Pi 3 reserves the memory above address 0x3F000000 for devices" and "AUXENB Register (0x7E21 5004)" to "0x3F00 0000 + 0x0021 5004" ? If my research are correct, the Rapsberry Pi 2 use 0x20xx xxxx as the base for physical addresses and 0x3Fxx xxxx for the Raspberry Pi 3. It is specified in the BCM2835 manual that you need to go from the bus address (which is based at 0x7Exx xxxx) to the physical address but maybe it is worth mentioning in the course ?
Also, why we can't use RXD0 and TXD0 for GPIO seems a bit obscure for me. I found in the documentation that we have UART0 (PL011 UART) and UART1 (Mini UART from Auxiliaries). IMHO, it could be nice to mention it in the course (maybe something like "we are using RX/TX D1 because instead of RX/TX D0 that are mapped to the other UART) and maybe give a reference to https://www.raspberrypi.org/documentation/configuration/uart.md for basic differences ?

s-matyukevich, 2018-06-25:
@T4g1 Thanks for your input!

I think this will be clear when I update the peripheral manual to the one mentioned in this https://github.com/s-matyukevich/raspberry-pi-os/issues/38 issue. You can download the new manual and read section `1.2.3 ARM physical addresses` - it should answer your question about memory calculations.
I will also update the tutorial to mention the difference between UART0 and UART1. If you want to see how the RPi OS works with UART1 you can check exercise  2 from lesson 1. We already have a few PRs with answers for that exercise.

T4g1, 2018-06-26:
No problem. I have not yet started any practical work (still gathering the necessary hardware) but i wanted to share what I found unclear. If it helps, I'll continue to do it. Do you have a preference for such communication or should I continue to use issue for that?

s-matyukevich, 2018-06-26:
I like when people provide their feedback - it really helps to improve the project. Though, unfortunately, I not always have time to address all issues quickly.  
Github issues work fine for such kind of communication.