s-matyukevich/raspberry-pi-os #70: [Lesson03] Exercises questions

bl4ckout31, 2018-06-30:
I have 2 questions about the exercises of Lesson03 that I'm currently trying to complete, respectively exercise 1 and 3.

1. Is the `local timer` the `Timer (ARM side)` of BCM2835 Peripherals manual p.196 ?

2. Is the `ARM generic timer` mentioned [here](https://github.com/s-matyukevich/raspberry-pi-os/issues/8#issuecomment-395319084) the one mentioned in my previous question or `The Generic Timer in AArch64 state` of Aarch64-Reference-Manual p. 2321 ?

s-matyukevich, 2018-07-01:
@bl4ckout31, The local timer described in the `BCM2835 Peripherals manual` was designed for the old single-core Raspberry Pis - it may still work because of compatibility guarantees. However, I recommend you to use the local ARM timer described in the [BCM2836 ARM-local peripherals](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2836/QA7_rev3.4.pdf) manual. (I mentioned it [here](https://github.com/s-matyukevich/raspberry-pi-os/blob/master/docs/lesson03/linux/interrupt_controllers.md) but I will also add this link to the exercise definition)
I am not very familiar with the `Generic timer` mentioned in the `AArch64-Reference-Manual p. 2321`  but I will try to investigate what exactly it does and how it is different from the ARM local timer.

closed 2018-07-05 by s-matyukevich

f0rm2l1n, 2019-10-08:
@bl4ckout31 
Hi there, I want to ask some questions about exercise1 :D
I want to emulate the timer exception handling in QEMU while the standard timer seems not supported (https://www.raspberrypi.org/forums/viewtopic.php?t=195565)

So can you tell me how you did it? Did you use a real machine or something else?
thanks a lot!

bl4ckout31, 2019-10-09:
@f0rm2l1n I used the RPi 3 local timer for the real hardware and the ARM generic timer for QEMU.
I don't remember much to be honest, that's what a gathered from reading my code.
Maybe the local timer is working now with QEMU 4.x, I'm gonna try.

bl4ckout31, 2019-10-09:
@f0rm2l1n After some testing, it looks like QEMU now implement the RPi 3 local timer with the same behavior, at least on version 4.1.0. For this exercice, the same code apart from `boot.S` and `linker.ld` should work on both the real hardware and QEMU.

PR #72 (2018-07-05): exercises/bl4ckout31: add lesson03 answers
I got the generic timer to work on my machine (qemu 2.12.0).
Feel free to test it before merging and close #70.