s-matyukevich/raspberry-pi-os #75: Questions about Mini UART in lesson 1

paulelong, 2018-07-09:
I've successfully built the first lesson and run it on my Raspberry PI.  But as I was reading through, two questions related to the UART configuration came up.  Perhaps they can be clarified in the doc.

1. The lesson states that we want to use the Mini UART and that pins 14 and 15 have the TDX1 and RDX1 functions.  So after doing the lesson I think I understand that  TDX0 maps to the PL11 UART and TDX1 maps to the Mini UART, but this was not clear when I read through the first time.  Also the Broadcom documentation doesn't explicitly state this either.

2. Related to #1, it mentions that "This means that if we select alternative function 5, they will use Mini UART."  But the code shows that a 2 is moved into GPFSEL1, which I thought would set alt2.  Instead I expect a 5 to select alt5.  Assuming I'm missing something, because 2 does seem to work, I think it an explanation why 2 is set instead of 5 would make it clearer.

I think this is such a great resource, and agree completely with your introduction.  Having a simple starting place to understand how an OS works is extremely useful to me.  Thanks for all the work so far, and look forward to the current and future lessons.

Thanks,

Paul

bl4ckout31, 2018-07-10:
1. It is not said in the lesson but it is explicitly stated in the Broadcom documentation. There is a `Special function legend` table p. 103-105 where every name from the `Alternative Function Assignement` table above is briefly explained with the corresponding section in the document. Granted, it is not really clear in the lesson and a reference to it would avoid some confusions.

2. The table describing the `GPFSEL1` register in lesson 1 has your answer. There you can see that the 3 bits function selection is not in natural order: `100` is alternate function 0, `010` is alternate function 5, …

s-matyukevich, 2018-07-10:
@bl4ckout31  - thanks for the answer!
@paulelong  - I'll add more detailed descriptions for both your questions in the tutorial.

paulelong, 2018-07-12:
For #2, I now see that the alt functions aren't consecutive nor in order.  Might be useful to call this out.  It required closer examination of the table to notice the peculiar order. For reference using decimal values alt0 = 4, alt1 = 5, alt2 = 6, alt3 = 7, alt4 = 3, alt5 = 2.  Seems like madness!  Though I wonder if there's a reason or story behind it :)

paulelong, 2018-07-19:
Thanks @s-matyukevich, I can help review any change you make.

Thanks #bl4ckout31, I suppose I was expecting the chart to mention Mini UART and PL011 UART, rather than UART 0 and UART 1.  I think reading the UART section I can then figure out the difference, and perhaps that in itself is a good exercise in reading technical documentation :)