s-matyukevich/raspberry-pi-os #76: Baud rate and baud rate register calculation

mitjans, 2018-07-10:
In [Introducing RPi OS, or bare-metal "Hello, World!"](https://github.com/s-matyukevich/raspberry-pi-os/blob/master/docs/lesson01/rpi-os.md#initializing-the-mini-uart) the formula given to calculate baud rate is:

```
baudrate = system_clock_freq / (8 * ( baudrate_reg + 1 ))
```

The `system_clock_freq` is 250MHz, and using 270 for the `baudrate_reg`, the value given is ~115314, and not 115200. If I open a screen using 115314 for the baud rate, I get gibberish.

Why is that? How can I calculate correctly the `baudrate_reg` given any `baudrate` value for exercise 1, lesson 01?

Thanks

s-matyukevich, 2018-07-10:
@mitjans I guess that not all baud rates are supported by the Mini UART, and the result of the baud rate calculation is rounded somehow. Though I wasn't able to find any proof in the documentation.

H-4ND-H, 2018-07-12:
In UART, baudrate need not to be a exact value, some tolerance in agreed upon baudrate is acceptable(about ~2% is allowed), so value 270 for 115200 communication is alright.
@mitjans So That's why communication is working with 270 and must also work with 115314 baudrate  too,  I wonder what method you use to open a UART session with non-standard baudrate, but I think your main issue must be something with driver not allowing you to choose non-standard baudrate, 
knowing your OS and method and driver you use may help finding where the issue lies,
Regards