s-matyukevich/raspberry-pi-os #78: I got a problem about using docker ti build

LBBym, 2018-07-13:
Hello!
I‘m using dockertools in Winsows10 . 
the command in my build.bat is
`docker run --rm -v %cd%:/app -w /app smatyukevich/raspberry-pi-os-builder make %1`
I think it's just the same as the command in the repo. But everytime I try to run `./build.bat` in powershell, it would say `docker: Error response from daemon: invalid mode: /app.
See 'docker run --help'.` I've already search in the internet and still work it out. Could anybody help me out? Thanks a lot!

s-matyukevich, 2018-07-13:
@LBBym I think that %cd% works only in Windows Command Line (cmd), but not in PowerShell. Can you try replacing it with `$(pwd)`?  You can also try to run `build.bat` from cmd. Check [this](https://stackoverflow.com/questions/41485217/mount-current-directory-as-volume-in-docker-on-windows-10) question for reference.

LBBym, 2018-07-14:
Thanks a lot !
I've fixed it

closed 2018-07-14 by LBBym