s-matyukevich/raspberry-pi-os #97: Lesson 1 should add instructions to building a cross-compiler

dublevsky, 2018-09-11:
I keep running into an error with locating pthread.h while building gcc cross-compiler for the target from lesson 1 (aarch64-linux-gnu) even though I've successfully built x86 and x64 compilers before. I think I'm not the only one who faced this problem and would love a detailed guide from the author himself.

P.S. I followed [this](https://wiki.osdev.org/GCC_Cross-Compiler) guide/

s-matyukevich, 2018-09-11:
@dublevsky have you tried to build with docker?

dublevsky, 2018-09-11:
No, I have no experience with docker.

s-matyukevich, 2018-09-11:
I prefer to use docker because it provides you with a consistent cross-platform experience and I don't have to deal with different host/OS/distro specific issues. I can't guide you how to build the compiler, because I haven't done it by myself ([here](https://github.com/s-matyukevich/raspberry-pi-os/blob/master/Dockerfile#L3) you can see how I installed the compiler on ubuntu)

closed 2018-09-25 by s-matyukevich