s-matyukevich/raspberry-pi-os #99: Why are we compiling assembly files using GCC instead of GAS?

dublevsky, 2018-09-25:
Hi, wanted to know what's the reason in compiling .S files using GCC rather than GAS. Is it because of "-MMD" flag?

P. S. I think there should be some kind of IRC or Discord channel to ask these kinds of questions instead of polluting the "Issues".

s-matyukevich, 2018-09-26:
We need GCC because we want to reuse in the assembler code some macros defined in the header files. Otherwise, we will have to duplicate a lot of constants and define them separately for C and assembler code.

s-matyukevich, 2018-09-26:
@dublevsky I think I'll create a slack channel to simplify the communication.

s-matyukevich, 2018-09-28:
I've added a link to the slack channel in the readme. Closing this issue now.

closed 2018-09-28 by s-matyukevich