s-matyukevich/raspberry-pi-os #106: Why use 0x3F000000 as base address?

riyadshauk, 2018-11-14:
It would be great if you could add this kind of explanation to the README.md, as the manual-page link that you referred us to was showing a different base address, and it didn't really click for me what was going on for a while... Here was a great/helpful analysis/explanation of what's going on, in simple: https://www.raspberrypi.org/forums/viewtopic.php?t=142439 .

s-matyukevich, 2018-11-15:
@riyadshauk in the  `BCM2837 ARM Peripherals manual` , section `1.2.3 ARM physical addresses` it says  the following: 
```
Physical addresses range from 0x3F000000 to 0x3FFFFFFF for peripherals. The
bus addresses for peripherals are set up to map onto the peripheral bus address range
starting at 0x7E000000. Thus a peripheral advertised here at bus address 0x7Ennnnnn is
available at physical address 0x3Fnnnnnn.
```
It looks correct to me. Maybe I should just emphasize the fact that we are using physical addresses rather than bus addresses in the tutorial. i'll try to add a little bit more explanations.

riyadshauk, 2018-11-15:
Yeah, sorry... I missed that detail in the manual. Yeah, exactly that suffices (that exact excerpt). It would be helpful if you spoon-fed us to look at that section, maybe? I'm a noob with electronics and am not accustomed to seeing such a big manual for just hardware. But maybe I'm expected to just be better at looking through the linked manual?

Maybe after this sentence in the Lesson 1 README,
"The meaning of each bit in each device register is described in the BCM2837 ARM Peripherals manual.",
you could add something like:
"Take a look at section `1.2.3 ARM physical addresses` in `BCM2837 ARM Peripherals manual` and the surrounding documentation for more context on why we use `0x3F000000` as a base address (even though `0x7E000000` is used throughout the manual)."

pointing it out in such a way might just save confusion/time for others... Though, I understand that a lot of things are important to look at, and highlighting one specific part of a document might make other parts seem less relevant/important, so idk...

In conclusion, it seems to me (personally) that revising that paragraph to include a specific reference to section `1.2.3` would be more helpful than not.

s-matyukevich, 2018-11-19:
Thanks @riyadshauk  for your feedback. I updated the lesson accordingly to your suggestion. https://github.com/s-matyukevich/raspberry-pi-os/commit/d4dd37b4fd736f87e57a9f1d3b66d3b460a4bb33

closed 2018-11-19 by s-matyukevich