s-matyukevich/raspberry-pi-os #114: getting error while running ./build.sh or ./build.bat docker command not found

chethanomkar, 2019-01-02:


chethanomkar, 2019-01-02:
Hi, actually when i try to run those files i'm getting the error like ./build.sh: line 3: docker: command not found. I earched in the google i couldn't find the solution.

s-matyukevich, 2019-01-02:
You need to install docker. You can find the link in the [Prerequisites](https://github.com/s-matyukevich/raspberry-pi-os/blob/master/docs/Prerequisites.md)

rockytriton, 2019-01-02:
you don't need docket to build it, i just ran make

s-matyukevich, 2019-01-03:
Yes, It will work without docker either if you have aarch64-linux-gnu toolchain installed on your machine.

chethanomkar, 2019-01-03:
Thank you for all your inputs.
I tried with make, by installing the gcc-aarch64-linux-gnu then the kernel image is generated now.

closed 2019-01-03 by s-matyukevich