s-matyukevich/raspberry-pi-os #118: RPI 3 with ARM v8 Boots into EL2 not EL3

justinarose, 2019-03-09:
When booting to Pi for lesson 2, it begins in EL2 not EL3. I believe this was a change. The RPI firmware now changes initial boot from EL3 to EL2.

s-matyukevich, 2019-03-11:
Thanks @justinarose  for reporting this. I'll check it out. In the meantime, you can use solutions of exercise 1 to see how you can jump from EL2 to EL1 [link]{https://github.com/s-matyukevich/raspberry-pi-os/tree/master/exercises/lesson02/1}

justinarose, 2019-03-11:
I think I may have found the underlying issue. If you forget the config.txt newer PIs automatically boot into EL2. When I updated that and firmware files, worked correctly.
Really love these tutorials and hope that the whole lesson plan can be finished! Extraordinarily helpful for my college RPi operating system class!

s-matyukevich, 2019-03-11:
Thanks, @justinarose! Closing this issue.

closed 2019-03-11 by s-matyukevich

charlestolley, 2021-04-28:
@justinarose could you clarify how you fixed this?