s-matyukevich/raspberry-pi-os #123: Put non-primary cores to sleep instead of endless loop

Benjoyo, 2019-05-08:
I just finished lesson01 and was wondering if the `proc_hang` endless loop that all non-primary cores go into wouldn't put 100% load on these three cores?

I was looking into the ARM documentation trying to find an instruction that basically disables a core and found `wfe` (wait for event) which will completely disable the clock for that core until an interrupt occurs. I think that is a bit nicer than a loop.

So I suggest:

```
proc_hang: 
	wfe
	b 	proc_hang
```

s-matyukevich, 2019-05-08:
Thanks, @Benjoyo ! This makes sense. I'll update the code.

Benjoyo, 2019-05-08:
@s-matyukevich btw thank you so much for this amazing resource, it's simple enough to understand everything and functional enough to motivate - and everything is explained in much more detail than elsewhere.
I hope there will be more lessons in the future, keep it up :)

s-matyukevich, 2019-05-10:
Thanks, @Benjoyo ! I made a big pause but recently I resumed the work, so hopefully, I'll be able to finish lesson 7 soon.