s-matyukevich/raspberry-pi-os #127: Will this tutorial work on Raspberry Pi 4?

ZzzGin, 2019-07-03:
I've read the introduction of this tutorial and I really think this is what I want to start my journey of Linux kernel. Thank you for this material.
For now, the 4th gen of raspberry pi is out so I am not sure if this repo works on the new Pi (whose SoC is **Broadcom BCM2711B0 quad-core A72 (ARMv8-A) 64-bit @ 1.5GHz)**. 
If it won't work, I decide to buy the one mentioned in the Introduction.

s-matyukevich, 2019-07-04:
**Short answer**: No, we have to update the code to make it work on the RPi 4b.

**Long answer:**
Looks like  BCM2711B0 has different peripheral Base address (0xFE000000)  [link](https://www.raspberrypi.org/forums/viewtopic.php?t=244479&p=1491494)  this means that at least  [this](https://github.com/s-matyukevich/raspberry-pi-os/blob/master/src/lesson01/include/peripherals/base.h#L4) line has to be updated. Based on what I saw in the [bcm2711-rpi-4-b.dts](https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/bcm2711-rpi-4-b.dts) looks like mini UART, interrupt controller and timer should work. MMU should also work fine as well. So hopefully base address should be the only change we have to make.

ZzzGin, 2019-07-04:
@s-matyukevich thanks for replying. 
I will use a 3 gen for this tutorial. And I will try it on the new pi when I get a 4th gen, since it is out of stock on its official website.

closed 2019-07-04 by ZzzGin

vikash876, 2019-08-14:
No luck with above changes (by changing pbase to 0xFE000000). No output on serial console. Anyone had any luck?

Also one more question, regarding changes in config.txt. Do we need to replace older config.txt so that the new one has only two lines:
kernel_old=1
disable_commandline_tags=1

Or we need to add above two lines below our existing config.txt? Don't we need enable_uart=1?

vikash876, 2019-08-14:
Pheww... Working now!! :) So config.txt needs to have just "arm_64bit=1" and nothing else. Digged into this thread: https://www.raspberrypi.org/forums/viewtopic.php?t=244479&p=1491494 to find this info.

rockytriton, 2019-08-16:
Hey, so you were able to get the UART working?  I tried the same things with my pi 4 and no luck.  Can you post the contents of your config.txt?  Also did you have to download the 2711 dtb file and fixup4.dat and start4.elf?

vikash876, 2019-08-18:
Hi rockytriton,

Here is my forked version which works with rapsberry pi-4 : git@github.com:vikash876/raspberry-pi-os.git
Steps to build:
1. git clone git@github.com:vikash876/raspberry-pi-os.git
2. build the lesson01 
3. copy config.txt from lesson01/src/ and overwrite on your sd card
4. delete kernel7.elf from your sd  and copy kernel8.elf to your sd card.

Also I have your answers here:
rockytriton: so you were able to get the UART working?
ans: Yes, I am using USB to TTL cable made by adafriut(https://www.adafruit.com/product/954).Any similar cable should work. I can see the print "Hello, world!"
rockytriton: Can you post the contents of your config.txt?
ans: please check https://github.com/vikash876/raspberry-pi-os/blob/master/src/lesson01/src/config.txt
rockytriton: Also did you have to download the 2711 dtb file and fixup4.dat and start4.elf?
ans: I did following:
1. Flash the sd card with raspbian (either lite or full)
2. delete the kernel7.elf
3. copy and replace the config.txt from my forked version lesson1/src/config.txt (as above link)
4. copy the kernel8.elf
5. Boot the raspberry pi-4 with modified sd card and and set serial baud rate to 115200 in console.
6. Observe "Hello, world!" on screen :))

Changes from s-matyukevich version
1. Changed config.txt
2. Changed Base address (to: 0xFE000000)
3. Set the new baudrate (to: 541. since new GPU clock rate is 500MHZ, replaced this value in the formula)

hope this helps.

rockytriton, 2019-08-18:
Thanks for all the details.  I was pretty much doing the exact same things, except the 541 change.  However I figured out what my issue really was...   With the pi3 I would always use the serial cable's red cable to power the pi, so I didn't have to have its power source plugged in too.  It seems that with the pi4 maybe that's not enough power to boot it.  I put a usb cable into the pi4 to boot it and it's now working finally.

vikash876, 2019-08-18:
Glad to hear. Yes I always tried with usb c powered. A 3amp power input is must I suppose.

AlexGatz, 2020-02-02:
Thank you! Somehow I looked over this. I had just realized I was looking at the datasheet from the bcm2835 instead of the bcm2711.

AlexGatz, 2020-02-02:
Hello, wo0>���p>0 is my output lol

danieltomch17, 2020-06-06:
> Changes from s-matyukevich version
> 
> 1. Changed config.txt
> 2. Changed Base address (to: 0xFE000000)
> 3. Set the new baudrate (to: 541. since new GPU clock rate is 500MHZ, replaced this value in the formula)

I have done all the above. The “Hello World” is printed just as expected, but the the kernel_main gets into a loop where it reads from the uart and prints back whatever is read - but in my case it prints gibberish. Any ideas on how to fix that problem?

bcoppens, 2020-07-07:
> I have done all the above. The “Hello World” is printed just as expected, but the the kernel_main gets into a loop where it reads from the uart and prints back whatever is read - but in my case it prints gibberish. Any ideas on how to fix that problem?

You can fix this by adding `enable_uart=1` to your config.txt. I presume that otherwise the VPU dynamically scales down the core frequency after booting, so that the clock is only correct while printing the first couple of characters before it scales itself down; and that the `enable_uart=1` prevents this. (This is not 100% clear from the documentation that this would be so on the Pi 4B; I've submitted an issue on the Raspberry Documentation page about this: https://github.com/raspberrypi/documentation/issues/1614).

It's also not clear why some people (such as myself) need the `enable_uart=1` to get things working, and others don't... (Maybe a different firmware version?)

AbramQuerashi, 2020-08-02:
Hi All,

From all the above comments it looks like I can purchase a RPi4 4GB memory board and this project is build able and can run on RPi4 ?

-abram

jyotimahapatra, 2020-09-07:
https://github.com/vikash876/raspberry-pi-os/commit/b4304230575e196ac716c46c39c13bf4095e84a4 was pretty much everything i needed to make it work. Thanks a lot!