s-matyukevich/raspberry-pi-os #130: Printing to real screen

robitec97, 2019-08-08:
I was trying to implement printing to real screen using the already made code found on this repo: https://jsandler18.github.io/tutorial/hdmi.html mayeb someone can help me with this!

rockytriton, 2019-08-08:
I played around with this same thing, here's a repo i have which is mostly based on what I learned from this repo.  I added a video.c which would write things to the screen using the HDMI output:  https://github.com/rockytriton/DSOS/tree/master/src

The problem here is that this uses the CPU to draw on the screen, which is ungodly slow.  If you have a whole screen of text it takes several seconds to draw it.  The only way to fast write to the screen is using the GPU, which I couldn't find any documentation on.

mmaret, 2019-09-05:
Hi, maybe you can get some information from here : https://www.cl.cam.ac.uk/projects/raspberrypi/tutorials/os/screen01.html

Jesus8000, 2021-01-16:
hi, @robitec97 i have the same problem with the jsanders OS. Did u get it to work?

robitec97, 2021-01-28:
@Jesus8000 Unfortunately, no.

closed 2021-04-08 by robitec97