s-matyukevich/raspberry-pi-os #133: Trial on RPi4

amulyanv, 2019-09-04:
Hello!

I am very new to RPi and building an OS. I got a brand new RPi4 precisely to learn these concepts and your project fit my idea perfectly. Unfortunately, I tried to follow tutorial 1 and failed. I just get a coloured screen and no response on the serial. I understand coloured screen means the kernel did not boot correctly. So, could you give me some pointers on what could be different with RPi4 and in which direction I should go?

Thanks!

s-matyukevich, 2019-09-04:
@amulyanv - you can check this issue for details https://github.com/s-matyukevich/raspberry-pi-os/issues/127

amulyanv, 2019-09-04:
Thanks so much!! it works now. The base address was the problem :D

closed 2019-09-04 by amulyanv