s-matyukevich/raspberry-pi-os #161: Gibberish in terminal emulator on Lesson 1

scdrumz, 2019-12-14:
Definitely related to #127, but I didn't want to reopen an older thread.

I've completed lesson 1 on my pi 4b, and made the modifications mentioned in the above thread, specifically:

Replaced the contents of config.txt with:
```
arm64bit=1
```

and then modified src/mini_uart.c to change the baud rate:
```
    put32(AUX_MU_BAUD_REG,541);
```

I've verified my code against both this repo and @vikash876, and it appears to match up.

But, using putty, I just get gibberish output. I tried the serial monitor in my arduino app to see if there's a difference, and it's a little less muddled:

[image]

But, it _is_ still incorrect. Everything I'm seeing currently indicates this could just be attributed to the mini uart being tied to the GPU clock as compared to the PL011 uart.  Would that seem reasonable? I'm inclined to move forward as-is if so -- I see the same behaviour when I tried to set to 9600 baud.

Is there anything else obvious I might be missing?

andre-richter, 2019-12-14:
I had exactly the same problem. I spent a whole day trying to understand what is happening. At one point I hooked the UART lines to a logic analyzer and saw that while the first few characters are okay, later ones had different bit times. Hence, I also concluded that it must be some sort of DVFS going on, and in fact it seems to be the case: https://www.raspberrypi.org/documentation/configuration/uart.md

However, none of the config options to fix the frequency worked for me on the Pi4. For my project (https://github.com/rust-embedded/rust-raspi3-OS-tutorials/tree/master/06_drivers_gpio_uart) I eventually ditched the mini UART altogether. Using the `init_uart_clock`  config.txt option (description in the link) makes the PL011 UART setup as easy as the miniUART setup, because you don't need the mailbox to change frequency settings.

rockytriton, 2019-12-14:
If you are using the pi 4, be sure not to try to power it with your serial cable.  I made the mistake and spent tons of time trying to debug it, but once I plugged in a usbc to power it everything was fine.  it needs more amps than the usb serial cable will provide.

andre-richter, 2019-12-14:
For me, Pi4 works with USB serial cable only (only PL011 UART, as stated above, though).

scdrumz, 2019-12-15:
Cool, this all makes sense. I'll start with the exercise to enable the
PL011 and go from there. I definitely am using the USB power, I spent my
first night playing with the wiring getting all that down :).  And, I
_have_ verified the serial adapter itself on my arduino board, so I know
that bit's good.  Glad to hear others have run into this as well.

AlexGatz, 2020-02-02:
I'm honestly lost here. Can someone explain how to setup PL011 in more detail? I did find [this](https://www.raspberrypi.org/forums/viewtopic.php?t=185907). I'm just not skilled enough in baremetal coding to have a clue as to what is needed to get this to work with these lessons. Feeling very discouraged after spending half a day to only get gibberish lol.

rockytriton, 2020-02-02:
if you are that lost, I would recommend this udemy class, it's about stm32, not raspberry pi, but it's arm based and teaches you how drivers are developed.  The instructor is Indian but very easily understandable:  https://www.udemy.com/course/mastering-microcontroller-with-peripheral-driver-development/

AlexGatz, 2020-02-02:
Thanks, I'll check it out. I did figure out what my issue was shortly after making that comment.