s-matyukevich/raspberry-pi-os #165: Lesson 4: scheduling of a new task from interrupt

Lowentwickler, 2020-01-04:
Lets pretend, timer interrupt occurs and we enter interrupt as one task and we want to exit it as new task (created during execution of the first one).

```
el1_irq:
	kernel_entry 
	bl	handle_irq
	kernel_exit
```

`kernel_entry` macro will store registers including `elr_el1` and `spsr_el1`, `handle_irq` will perform task switching, but `kernel_exit` macro will try to load registers from stack, but they were never written into it, weren't they?

And the second question, regarding kernel_entry and kernel_exit macro: if we perform task switching inside interrupt, it seems we store and restore x19-x30 registers twice:
1. in cpu_switch_to
2. in el1_irq
Or I misunderstand something?

s-matyukevich, 2020-01-07:
@Lowentwickler 
1. The first time the kernel runs a task `kernel_exit` macro isn't executed - we jump directly to the `ret_from_fork` function instead, and no registers are restored. In lesson4 we don't need to run `kernel_exit` in this case, because all our tasks are running using the same exception level. This, however, will be fixed in lesson5 [link](https://github.com/s-matyukevich/raspberry-pi-os/blob/master/src/lesson05/src/entry.S#L188)
2. Yes, we store them twice.  You can think about it in the following way: `kernel_entry ` saves the actual state of the task that is preserved between interrupt calls. Next interrupt handler can potentially update some of the x19-x30 registers, and `cpu_switch_to` saves them as well. We also use `cpu_switch_to` to set x19 and x20 registers for the [new tasks](https://github.com/s-matyukevich/raspberry-pi-os/blob/master/src/lesson04/src/fork.c#L18)

Lowentwickler, 2020-01-21:
But in this case you have a task running inside interrupt. Is it normal?

s-matyukevich, 2020-01-21:
Yes, that is normal, as long as you don't forget to reenable interrupts and you don't have to switch EL on exception return.

Lowentwickler, 2020-02-20:
How does the case handled, when `_schedule`, called from Interrupt (EL1) selects non new task from userspace (EL0) to be executed? It is more about lesson 5. It seems `cpu_switch_to` will switch to userspace task context, but in fact it will run with EL1 privileges. Or I miss something again?

s-matyukevich, 2020-02-20:
In this case `cpu_switch_to` won't be the place were privileges are dropped. After `cpu_switch_to` finishes we will be running in EL1 mode and the executions jumps [here](https://github.com/s-matyukevich/raspberry-pi-os/blob/master/src/lesson05/src/sched.c#L67) (this is the place were newly selected task stopped last time) Next, the execution goes [here](https://github.com/s-matyukevich/raspberry-pi-os/blob/master/src/lesson05/src/entry.S#L155) and this is the place were privileges are dropped and we go back to EL0.

Lowentwickler, 2020-02-20:
I see, makes sense: user space task is being switched only inside the interrupt, so recovered in the same interrupt as well and returns to his real EL0 duties only after eret.

Uolsen, 2022-11-28:
> Yes, that is normal, as long as you don't forget to reenable interrupts and you don't have to switch EL on exception return.

@s-matyukevich Yes but it doesn't seem that it's working. When i'm in "IRQ mode" I'm not enabled to make second interrupt. Is there some code that reenabled interrupt, DAIF register didn't do nothing