s-matyukevich/raspberry-pi-os #166: Lesson01 not working with raspberry pi 4 model b.

AlexGatz, 2020-02-02:
Hello! I realize this has not been updated for quite a while now. I'm just very interested in this project and I started working through lesson01.

Setup:
- Arch Linux
- Raspberry Pi 4 Model B 4gb [link](https://www.raspberrypi.org/products/raspberry-pi-4-model-b/)
- Uart to USB from adafruit [link](https://www.adafruit.com/product/954)

Things i've done:
- Double checked that the uart is connected properly.
- Identified that the PBASE is now 0x7E000000 instead of 0x3F000000, made that change (Was getting an error in dmesg and tty was not working at all). Tried both anyway.
- Found the datasheet for the new broadcom chip [BCM2835](https://www.raspberrypi.org/app/uploads/2012/02/BCM2835-ARM-Peripherals.pdf) and identified that it is still 250Mhz so the baudrate should be accurate.

It's probably something obvious to someone. I'm just literally brand new to this. Had I known it was this specific to pi3 I would have just bought one of those :(

s-matyukevich, 2020-02-02:
@AlexGatz  you can check [this](https://github.com/s-matyukevich/raspberry-pi-os/issues/127) issue for details.

AlexGatz, 2020-02-02:
So sorry, overlooked that issue. I'll read it right now. Thank you.

AlexGatz, 2020-02-02:
Its working. Had to use `init_uart_clock` to enable PL011 UART which apparently has a fixed clock rate of `init_uart_clock=48000000`. Thanks again.

closed 2020-02-02 by AlexGatz