s-matyukevich/raspberry-pi-os #168: One issue that confused me in lesson 6, about link address

liuye0331, 2020-02-25:
This is a great tutorial, I learned a lot of concepts about linux in this site. But one issue confused me in lesson 6:
In linker.ld, the third line set link address to 0xffff000000000000, and in function kernel_process invoke move_to_user_mode, inside which invoke allocate_user_page. The second parameter of allocate_user_page is 0. I think this means that in user space, the address is from 0 to 4k, but the link address is above 0xffff000000000000. How this could work normally?

s-matyukevich, 2020-02-25:
This works because compiler only uses relative addresses when compiling `user.c` and `user_sys.S` (those are 2 files that are copied to the new page in the [move_to_user_mode](https://github.com/s-matyukevich/raspberry-pi-os/blob/master/src/lesson06/src/fork.c#L54) function ) So in this particular case we don't care what the link address is.

liuye0331, 2020-02-26:
Thanks for your reply. But why compiler only use relative addresses when compiling user. c and user_sys. S.  I checked the makefile, no special options found.  Did I miss some thing?

s-matyukevich, 2020-02-27:
Because this is the default behavior: compiler always generates only relative addresses, unless it is not possible or you tell him otherwise