s-matyukevich/raspberry-pi-os #180: Remove requirement for kernel_old?

chunky125, 2020-03-29:
Not really an issue - more of a question!  I'm trying to get the lessons to boot without kernel_old in the config.txt file.

So far, I've adjusted the linking script with ". = 0x80000;" to drive the parts to the right place, and a bit of digging in the ELF output with GDB before the binary is generated suggest things have moved, but I get Lesson 1 to boot OK, and Lesson 2 onwards won't.

Any ideas on why this is?  I'm asking because I want to see if the DTB address is passed correctly without kernel_old?

s-matyukevich, 2020-04-10:
No really sure, but my guess is that this line https://github.com/s-matyukevich/raspberry-pi-os/blob/master/src/lesson02/src/boot.S#L30 doesn't work, as relative addresses in the boot.S no longer match absolute addresses.

chunky125, 2020-04-11:
Perhaps, I'm going to try and get lesson 2 running without kernel_old, I think it might be because the bootloader puts the ARM core into EL3 without kernel_old, rather than EL2.  I'll put a push in when I get it working.

keur, 2020-11-23:
@chunky125 You are correct, the linking is fine. It's that it starts at exception level 2, so you can't use registers only available in exception level 3.

[link_lesson_2.txt](https://github.com/s-matyukevich/raspberry-pi-os/files/5586369/link_lesson_2.txt)

rockytriton, 2020-11-23:
I have a video about fixing this:
https://www.youtube.com/watch?v=zRhMutscRL0&list=PLVxiWMqQvhg9FCteL7I0aohj1_YiUx1x8&index=6

So it will boot you into EL 3 instead of EL 2