s-matyukevich/raspberry-pi-os #182: Difference in rpi-os.md -- Should we "Clear FIFO"?

leoleoasd, 2020-04-04:
There isn't a "Clear FIFO" line at [here](https://github.com/s-matyukevich/raspberry-pi-os/blame/master/docs/lesson01/rpi-os.md#L295),  but it showed up [here](https://github.com/s-matyukevich/raspberry-pi-os/blame/master/docs/lesson01/rpi-os.md#L370). Which is correct?

Also, I don't know how to make a link with a line number with a markdown file ( which can be done by pressing the line number in other files ), so I'm using the "git blame" to generate a link with a line number.

s-matyukevich, 2020-04-10:
It is optional, as FIFO is already empty after start, the actual code doesn't use this line https://github.com/s-matyukevich/raspberry-pi-os/blob/master/src/lesson01/src/mini_uart.c#L52

leoleoasd, 2020-04-13:
So should we remove the "Clear fifo" line cause it is confusing?

s-matyukevich, 2020-04-14:
Yes, [fixed](https://github.com/s-matyukevich/raspberry-pi-os/commit/8e15715d73cd2dbd247f2fc55f374affd856bd96)

closed 2020-04-14 by s-matyukevich