s-matyukevich/raspberry-pi-os #184: how can i run it on Ri

figozhang, 2020-04-15:
i am using PI 3B+, i have a USB-Uart cable, and i install the latest "Raspbian Buster Lite" image on SD card, and i can see the Linux boot message on Uart. The Raspbian os is : https://downloads.raspberrypi.org/raspbian_lite_latest, it is build 2020-02.

i put our lesson01 image into SD card, replace kernel8.imge, but i canot see any output on Uart. what is problem?

boringangel, 2020-04-17:
Did you delete the other kernal*.img's, and change the config.txt?

stefanJi, 2020-04-30:
Do you change config.txt as follow ?

```
kernel_old=1
disable_commandline_tags=1
```

echowwww, 2020-04-30:
I had the same problem. 
I found that the output comes up very shortly after the device is powered on. So please try to open an emulator session, then power on the rasp pi.

sjtugjy, 2020-09-12:
I meet the same problem. Is there any progress? @figozhang 
I will be appreciated for any help.

obstriker, 2020-10-03:
I meet the same issue with Rpi 3B.
I did all of the above.

Discenna, 2020-12-30:
I've been able to run lessons 1-6 (not exercises) on my Pi 3B+.

After a fresh install of the Raspberry Pi OS on a SD card, using the Raspberry Pi Imager;

I can run lesson01 with the following changes:
1) in config.txt (only the below)
```
disable_commandline_tags=1
arm_64bit=1
```
2) and due to the lack of 'kernel_old' I also made the changes mentioned in [https://github.com/s-matyukevich/raspberry-pi-os/issues/180](url)
3) replace kernel8.img (in the SD card) with kernel8.img compiled from the modified lesson01 (modified due to (2))

My reasoning for not using 'kernel_old' was mostly due to preference, however, when testing lesson01 with 'kernel_old' in addition to (1) (minus (2)), I was unable to successfully display 'Hello World', although I'm unable to make any assertions as to why this may be (be it either a local issue or otherwise).