s-matyukevich/raspberry-pi-os #187: Is it okay to use ssh to substitute serial console?

BDHU, 2020-04-25:
Currently I'm stuck with setting up serial console because for some reason it won't take any of my keyboard inputs once it prompts me with the login screen. Is it safe to just use ssh? Since it's an OS level work I'm not sure if there are any later projects that might potential mess with the network setting. Or is serial console absolutely necessary?

stefanJi, 2020-05-07:
You could use *qemu* to run this project. By the way, linux based system will better to play with this project.

rmtonkavich, 2020-10-05:
If you think about your question and what you are trying to accomplish, you should be able to understand that trying to log in using openSSH is not an option. First you removed the Existing OS and Replaced it with Your Own OS. Next the OS you are installing is Really Bare Bones. There is no IP Address Available. As there is also No Standard "C" Libraries installed for things like (stdio) which SSH requires. Even if you were to plug in an Ethernet Cable as there is No Network Manager Available. In some respects the OS at this stage doesn't even Understand what a Terminal does?

The Serial Connection in my mind establishes a Modem Like Connection. You agree on the protocol and then send a String to Transmit to the RPi. I am not sure how you could get QEMU established at this particular point in the Lessons. Please don't be offended by my answer, sometimes we are looking for a solution that just does not exist. Respectfully Robert Tonkavich