s-matyukevich/raspberry-pi-os #191: config.txt

danieltomch17, 2020-05-16:
Hi,
I am reading lesson 1.
Didn’t quite understand how the config.txt comes into play. When is it used by the Pi and where does it read it from.
Can anyone please explain?
Thanks.

FedericoPonzi, 2020-05-17:
These specially-named files are recognized by the Raspberry Pi’s GPU on boot-up and used to configure and boostrap the system. bootcode.bin is the GPU’s first-stage bootloader. Its primary job is to load start.elf, the GPU’s second-stage bootloader. start.elf initializes the ARM CPU, configuring it as indicated in config.txt, loads kernel8.img into memory, and instructs the CPU to start executing the newly loaded code from kernel8.img.

danieltomch17, 2020-05-19:
Thanks a lot !
I don’t have an actual board (I’m running an emulator) so it might be the reason i don’t fully understand. I just drop those files on my sd card and the GPU will do the rest?