s-matyukevich/raspberry-pi-os #194: Raspberry Pi 4 Model B

jmielkeway, 2020-06-13:
I have been attempting to run lesson01 but with a Raspberry Pi 4 Model B. Based on [this](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0.pdf) peripheral manual for the BCM2711, I have made the following changes...

Changed
`#define PBASE 0x3F000000`
To be
`#define PBASE 0x47E000000`

Changed
`#define GPPUD           (PBASE+0x00200094)
#define GPPUDCLK0       (PBASE+0x00200098)`
To be
`#define GPIO_PUP_PDN_CNTRL_REG0 (PBASE+0x002000E4)`

And finally replaced
`	put32(GPPUD,0);
	delay(150);
	put32(GPPUDCLK0,(1<<14)|(1<<15));
	delay(150);
	put32(GPPUDCLK0,0);`
With
`        selector = get32(GPIO_PUP_PDN_CNTRL_REG0);
        selector &= 0x0FFFFFFF;
        put32(GPIO_PUP_PDN_CNTRL_REG0,selector);`

I used the raspberry pi image loaded to first image the card and tested the TTL connection, and it works well. After loading the kernel8.img and editing config.txt, however, there is no sign of life on the console.

Any thoughts on what I am missing?

Thank you!

jmielkeway, 2020-06-13:
It seems there is a "low peripherals" mode that may be configured, so I changed the base address to be 
`#define PBASE 0xFE000000`
This makes the console come to life, but the "Hello World!" message is gibberish, and the charcters I type are not the same ones that show up on the console.

I will keep investigating

roversch, 2020-06-30:
Isn't your `PBASE` one digit too long? By the way, even with your changes, the console is still dead for me.

roversch, 2020-06-30:
The changes mentioned in #127 did actually fix my issues for RPi 4B

BrainyZombie, 2020-07-30:
In lesson 1, AUX_MU_LCR_REG is being set to 3, which is correct according to the broadcom docs for RPi 3. 

However, according to the [docs](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0.pdf) (Pg 18) for RPi 4b, it need only be set to 1, with bit 1 being Reserved.

Yet I get gibberish when I set it to 1 and it works perfectly when set to 3.

figozhang, 2020-10-25:
in BMC2711's doc， the GPIO base address is：0x215000，it is different with RI3.

Andre-Fonteles, 2021-02-02:
> It seems there is a "low peripherals" mode that may be configured, so I changed the base address to be
> `#define PBASE 0xFE000000`
> This makes the console come to life, but the "Hello World!" message [...]

I know this topic is kind of old, but you might need to change the baud rate in your mini_uart.c from 270 to 541 to take into consideration the different clocks. I managed to run up to lesson 4 (where I got stuck with what I believe are other PI 4's differences).

ddnoodle, 2021-02-04:
@Andre-Fonteles hi, I succeed in lesson 01 with USB_to-TTL serial but failed on lesson 02. From quick debug, looks like boot.S in lesson 02 does not work for Pi 4B model. It is great to know you have run to up to lesson 4. Can you share me what changes you have made in lesson 02 ~ 04 to make it running on raspberry pi 4B model?

rockytriton, 2021-02-04:
@ddnoodle i have some videos that show how to do this stuff on the pi 4:  https://www.youtube.com/watch?v=pd9AVmcRc6U&list=PLVxiWMqQvhg9FCteL7I0aohj1_YiUx1x8

ddnoodle, 2021-02-06:
@rockytriton Great work and video tutorial! Thanks for your guidance on this and I can make lesson 2 work on Pi 4B and I am going to dig into more lessons.

fasihrana, 2021-12-14:
I'm following these lessons but I am unable to get any meaningful output on putty. My code is here [raspi](/fasihrana/raspi).