s-matyukevich/raspberry-pi-os #202: Stuck on rainbow screen on boot in Lesson 1

ptath1, 2020-07-27:
I have successfully compiled the image and transferred onto an SD card that had Raspbian installed. I deleted all the other kernel images in the SD Card boot partition, keeping the rest of the partition untouched from the Raspbian install. 

I have a Raspberry Pi 3B+ and I'm assuming the only difference between 3 and 3B+ is the processor speed, so I made no modifications to the code when I compiled the image. 

However, on boot I am stuck on the rainbow screen and I've run out of ideas to fix it. Has anyone else run into this issue or have any suggestions to fix this issue?

rockytriton, 2020-07-28:
If you are seeing the rainbow screen then you are looking at the wrong thing.  None of these lessons touch the video part of the raspberry pi.  Did you follow the prereq stuff about hooking up a usb serial connection?

ptath1, 2020-07-28:
I have serial working on Raspbian. I used "screen" on Raspbian and PuTTY on Windows and I was able to echo characters back and forth. But I assume, the whole purpose of that exercise is just to test whether serial is actually working on the PI with the cable. 

The only reason I plugged in the HDMI to see the Pi's screen, was to check if there are any warnings or errors that show up as my terminal on PuTTY is empty and non-responsive. I am using a mico-usb to power the PI, and have the session on PuTTY open waiting for the "Hello World".

I don't need "enable_uart=1" in my config file to make this work do I?

VictorMion, 2020-09-25:
I'm having the same problem.
Am I missing something?

obstriker, 2020-10-02:
I'm having the same issue and out of ideas..

rockytriton, 2020-10-04:
hey all, check my response here, it may be related to your issues:

https://github.com/s-matyukevich/raspberry-pi-os/issues/206#issuecomment-703317337