s-matyukevich/raspberry-pi-os #206: Lession01, can not be booted up with config.txt

drymon, 2020-09-23:
Hi @s-matyukevich,
When place config.txt into SD card, I can not see "Hello, world!". But if I remove this file, the "Hello, world!" will be displayed.
Does anyone have the same issue?
Also when try to run exercise 3 (multiple cores), I can see message from only one core. 
Do I need any further configuration to activate 4 cores of Raspberry PI 3 B?
Thanks!

rockytriton, 2020-10-04:
Yes, I confirmed that this is an issue with the latest raspberry pi OS firmware.  So if you use a newer raspberry pi OS as your base for the disk, it will actually not work.  You have to remove the:

kernel_old=1

and replace it with:

arm_64bit=1

With that configuration, it works for me now.  

If you use an older raspberry pi os firmware files, then you will need to use the same config.txt as in the tutorials.

MingliangLu, 2020-10-05:
> Yes, I confirmed that this is an issue with the latest raspberry pi OS firmware. So if you use a newer raspberry pi OS as your base for the disk, it will actually not work. You have to remove the: [...]

Thanks, It works for me.

drymon, 2020-10-05:
Thank @rockytriton ,
By setting arm_64bit=1 , it works for me in case of printing "Hello, world!".
But trying to activate multiple cores, it does not work.
I am checking template exercises: https://github.com/s-matyukevich/raspberry-pi-os/tree/master/exercises/lesson01/3/, but only message from one core such as "Hello, world! I am the Core 0" is printed out.

Could you please let me know the firmware version that you succeeded to activate all cores?

rockytriton, 2020-10-05:
@drymon I noticed that too, looking at the updated armstub code, it looks like they are putting the extra cores to sleep now and only letting the main core run the kernel code.  So it would be expecting the kernel to wake up the cores.  I saw someone else posted some info about waking up those cores here.  I haven't tried it yet myself:  https://github.com/s-matyukevich/raspberry-pi-os/issues/8#issuecomment-624115220

rockytriton, 2020-10-13:
FYI, I've created a video series which walks through the whole process for the first lesson.  My code is a bit different than Serge's but mostly it's the same as lesson 1.  It should help some of you who are having issues getting the serial part to work:  https://www.youtube.com/playlist?list=PLVxiWMqQvhg9FCteL7I0aohj1_YiUx1x8

drymon, 2020-10-14:
@rockytriton Thank you for your help. In order to make all the session code run properly, I have switched to the older firmware version and it work fine for me.
`commit afc71fd252cc947b50b3140cb3935af9ab0f260d (HEAD)
Author: popcornmix <popcornmix@gmail.com>
Date:   Wed Jun 27 21:31:19 2018 +0100

    kernel: Bump to 4.14.52
    
    kernel: Allo Katana DAC: Updated default values
    See: https://github.com/raspberrypi/linux/pull/2591
    
    kernel: ARM: bcm283x: Add missing interrupt for RNG block
    See: https://github.com/raspberrypi/linux/issues/2195
`

closed 2020-10-27 by drymon

vedangnaik, 2021-04-18:
Thanks for this! Using `arm_64bit=1` instead of `kernel_old=1` got lesson 1 to work on my Pi 3 Model B V1.2. 

I just wanted to add that, at least for lesson01, adding the `. = 0x80000` as mentioned [here](https://github.com/s-matyukevich/raspberry-pi-os/issues/180#issue-589874853) doens't do anything; in fact, the produced `.img` files are identical. This confused me a bit lol, so I just thought I'd mention it here.