s-matyukevich/raspberry-pi-os #207: [Lesson02] exercises - Switching EL in qemu

obstriker, 2020-10-05:
Did anyone manage to switch Exception levels in qemu?
I know when booting in qemu its on EL2 but even running the example or other solutions (Switching from EL2 -> EL1) didn't print anything.
Previous exercises were printing to screen but this one not printing for some reason.

X-141, 2020-12-27:
bump, I haven't found clear leads. There have been some pointers to possibly modifying QEMU's startup, however I have been looking for a easier solution first before I get into the weeds with that.

Because we are in EL2, we do not have access to sys registers that more privileged exception level (EL3) have access to. From what I've found and read, the instruction "msr  scr_el3, x0" causes a "prefetch abort" which will cause the system to hang-up. Commenting the instruction will allow the system to run, but of course we don't configure the Secure Config register.

robertfeliciano, 2022-07-16:
for anyone still looking (like I was lol). here is what i did. i can't attach .S files to this comment so i'll have to paste the code here.

```
main:
	ldr		x0, =SCTLR_VALUE_MMU_DISABLED
	msr 	sctlr_el1, x0
/* 
	ldr		x0, =HCR_VALUE
	msr 	hcr_el2, x0

	ldr 	x0, =SCR_VALUE
	msr 	scr_el3, x0
*/
	ldr		x0, =SPSR_VALUE
	msr 	spsr_el2, x0

	adr 	x0, el1_entry
	msr 	elr_el2, x0

	eret

```
so yeah, commenting out the hypervisor config and secure config registers, along with switching spsr_el3 and elr_el3 to their el2 counterparts fixes it. I now get to see "Exception level: 1" on printed on my terminal!

For anyone who wants to use my exact qemu command: 
`qemu-system-aarch64 -M raspi3 -kernel kernel8.img -serial null -serial stdio`

For anyone who wants to learn more about the new registers I mentioned here: 
spsr_el2 register on page 383
elr_el2 register on page 349 

hope this helps someone in my position.

lucas-infi, 2023-03-01:
Found the solution in this StackOverflow post: https://stackoverflow.com/questions/71006360/qemu-system-aarch64-always-entering-exception-level-2-and-hangs-when-switching-t

In short,
If you use qemu with the '-kernel' switch it will start in "Linux loader" mode, meaning it will try to start in EL2 is possible. Instead of using the `-kernel kernel8.img` flag, use `-device loader,file=./kernel8.img`, this will start qemu in EL1 mode.

jaiiarora, 2025-09-20:
Thanks, this helped a lot