s-matyukevich/raspberry-pi-os #216: lesson2 can't see Exception level: 3 on the screen

garlic8668, 2020-11-14:
When I run the lesson01, I can see "Hello World" displayed on my screen normally, but when I use printf in the lesson02, my screen does not display "Exception level: 3" or anything.

closed 2020-11-18 by garlic8668

subedisuman, 2021-01-31:
Similar issue for me. I am using RPI4B and am able to run lesson01 with no problem after applying modifications suggested in #127. When I run lesson02 with lesson01's `Boot.S`, I can `uart_send_string` `Hello World` as in lesson01 plus I could also print current EL by `uart_send`ing the `int el` as `char` but I was not able to see anything in screen with `printf` for the same. On the other hand, I was not even able to print anything on screen not even `Hello World` with `Boot.S` of lesson02.

ddnoodle, 2021-02-04:
I met the same issue on RPi 4B module.  Changing boot.S to the one in lesson 01 can see uart_send output, but no printf() output.
There must be something wrong in boot.S in lesson 02 on module RPi 4B. Who can help?

thanoskoutr, 2021-03-08:
I am currently working on a bare metal OS for the Raspberry Pi 4 and I also had trouble, getting lesson 2 to work on the Pi 4. Firstly, I was able to get the execution level and print it to the serial console (without changing the `boot.S`), and it displayed EL2. I believe if you have `kernel_old=1` set at the `config.txt` it might boots you at EL3, but I haven't checked it.
So I found that, with the new firmware, the kernel starts at EL2 and not EL3, as lesson 2 says. In order to change to EL1, I just removed the lines that change the EL3 registers and because the exception is taken at EL2, I save the `SPSR_VALUE` at the `spsr_el2` register and save the return address at `elr_el2`.

Because I haven't followed the lessons from the beggining, I have a different linker script and approach at booting. I start my kernel at `0x80000` and not at `0x0` memory address.

So I believe, if you delete the `kernel_old` option from `config.txt`, that loads the kernel at memory address `0x0` and place the kernel at `0x80000` through the linker script. It should work.

Also, I setup the stack manually at `0x80000`:
```
mov sp, #0x80000
```

Here is the changed `master` code, everything else in `boot.S` should be fine:
```
master:
	ldr x0, =SCTLR_VALUE_MMU_DISABLED
	msr sctlr_el1, x0

	ldr x0, =HCR_VALUE
	msr hcr_el2, x0

	ldr x0, =SPSR_VALUE
	msr spsr_el2, x0

	adr x0, el1_entry
	msr elr_el2, x0

	eret
```