s-matyukevich/raspberry-pi-os #217: [Lesson04] Raspberry Pi 4 Model B Make error

jeremypbrien, 2020-11-29:
After changing the PBASE to 0xFE000000 for the raspberry pi 4 model B, I get the following error when running make:

`build/boot_s.o: in function 'el1_entry':`
`(.text.boot+0x44): relocation truncated to fit: R_AARCH64_ADR_PREL_LO21 against symbol 'bss_end' defined in .bss section in build/kernel8.elf`
`make: *** [Makefile:30: kernel8.img] Error 1`

Lessons 1-3 work, but all other lessons are getting this error. Is there a workaround to make these lessons work for the raspberry pi 4 model B?

rockytriton, 2020-12-01:
Try changing this line:
https://github.com/s-matyukevich/raspberry-pi-os/blob/master/src/lesson04/src/boot.S#L41
It should be:
mov	sp, #LOW_MEMORY

Not sure why lesson04 changed it since lesson05 still has it using #LOW_MEMORY

maxstreitberger, 2021-07-10:
I got the same error, and I don't know what I should do. I already tried to add `#LOW_MEMORY`, but that doesn't change anything. It could be because `#LOW_MEMORY` is the same as `#(2 * SECTION_SIZE)`.

rhythm16, 2021-08-11:
Hi, I got the same error, and found that in my case the line `static unsigned short mem_map [ PAGING_PAGES ] = {0,};` in `mm.c` causes the bss section to be too large, exceeding the pc-relative addressing limit in the line `adr    x1, bss_end` in `boot.S`. I changed `mem_map`'s type to `uint8_t` and it compiled successfully. Caution this may not be the exact cause of your compilation error because I have set my `LOW_MEMORY` and `HIGH_MEMORY` differently (`0x40000000` to `0xFC000000`).

cuijier, 2022-05-11:
@rhythm16  I got the same error, and i haved tried all suggest, but still not work ,have anybody know the cause of error msg ?
(.text.boot+0x64): relocation truncated to fit: R_AARCH64_ADR_PREL_LO21 against symbol `bss_end'
timer.c:(.text+0x80): relocation truncated to fit: R_AARCH64_ADR_PREL_PG_HI21 against `.bss'
timer.c:(.text+0x90): relocation truncated to fit: R_AARCH64_ADR_PREL_PG_HI21 against `.bss'

cuijier, 2022-06-09:
About this problem,we can use `size`  cmd to check `build/*.o`  file  `.bss` section whether size is normal. for my problem, i found`mm_c.o` .bss section size is too big , which means uninit var is too big