s-matyukevich/raspberry-pi-os #218: The boot problem with Mac and ch340G chip usb2ttl.

yangyueren, 2020-12-03:
I have an problem with Mac and ch340G chip usb2ttl. Could someone help me?
When I use `screen /dev/cu.wchusbserial14130 115200` to connect to the original Pi OS, everything goes well. I can login and do everything.
[image]

But after I remove the kernel*.img in the boot and place the kernel8.img, as well as modify the config.txt to 
`kernel_old=1
disable_commandline_tags=1
enable_uart=1`,
I fail to see the hello world.
[image]

[image]

[image]
[image]

_Originally posted by @yangyueren in https://github.com/s-matyukevich/raspberry-pi-os/issues/43#issuecomment-737220879_

yangyueren, 2020-12-03:
I solved this problem by mofify the config.txt
replace `kernel_old=1`  with `arm_64bit=1`.
this problem is caused by the newer Pi os.

yangyueren, 2020-12-03:
I solved this problem by modify the config.txt
replace `kernel_old=1`  with `arm_64bit=1`.
this problem is caused by the newer Pi os.

rockytriton, 2020-12-03:
The new pios also puts you in a different exception level than the older ones, I have some code to revert it to how it was before so you can follow along with the lessons related to exception levels as well:  https://youtu.be/zRhMutscRL0

yangyueren, 2020-12-03:
> The new pios also puts you in a different exception level than the older ones, I have some code to revert it to how it was before so you can follow along with the lessons related to exception levels as well: https://youtu.be/zRhMutscRL0

Thanks for your quick reply. I will look into it.

closed 2020-12-03 by yangyueren

SunrShine, 2020-12-29:
同样的问题，是要在config当中加入arm_64bit=1吗

SunrShine, 2020-12-29:
[image]
when I open the putty and connect the raspberry-pi, there are Garbled codes as the picture shows. 
but if put the **bootcode_delay=5** in to config, using 5 seconds to open the putty and connect the raspberry-pi, it can get the right output.