s-matyukevich/raspberry-pi-os #221: [Lesson 1] UART sending garbage

jeremie-dautheribes, 2020-12-10:
I'm trying to test the lesson 1 but I'm only getting garbage from the mini-UART (see pic below).

[image]

It worked once but now I keep having the same garbage output.

I'm using screen with the following command:
```
sudo screen /dev/ttyUSB0 115200
```

I've already tested my USB-to-TTL converter on Raspbian and it works.

On Raspbian I had some issues with the baudrate so I tried to compile the kernel with baudrate = 9600 (adaptating `screen` command) but I still have the same output.

Here is my config.txt:
```
enable_uart=1
arm_64bit=1
disable_commandline_tags=1
```

I've set `arm_64bit=1` beacause I'm using a RPI 3B+ model and with `kernel_old` I don't have anything on the screen (even following the guide and removing all the kernel images).

jeremie-dautheribes, 2020-12-10:
Ok so when removing `enable_uart=1`, it works, whether with 9600 or 1152000 baudrate ...

I should have checked that before but I hope it will help someone 😄

closed 2020-12-10 by jeremie-dautheribes

thanoskoutr, 2021-03-02:
Thank you very much, you saved me from spending any more hours. I have left the config.txt from the Raspbian install and never thought that would be the problem.