s-matyukevich/raspberry-pi-os #225: lesson 01, How CPU find kernel_main in kernel.c from boot.S

xubenji, 2021-01-23:
Hi, Sergey Matyukevich. I am a beginner in developing operating systems. later on, I learned to develop OS in X86. Each time, in assembly files, if I want the CPU to find a certain function in .c files from assembly files, I have to include the keyword 'extern'. If I want the CPU to find function in assembly files, I have to include 'global'. But, in ARM assembly, I can't find any keyword to indicate the C function in .s files. So I  really want to find the answer. Thank you.

rockytriton, 2021-02-02:
Notice in this file, https://github.com/s-matyukevich/raspberry-pi-os/blob/master/src/lesson01/src/mm.S how the function memzero is defined, that can be called from C.

Andre-Fonteles, 2021-02-03:
I think xubenji question might  actually be about calling C functions from assembly files (not the other way around). Most of the time one would need to include the C header in the assembly file in order to call a function.

That said, I don't know how the boot.S file is able to call kernel_main without including any header that contains a reference to it. [https://github.com/s-matyukevich/raspberry-pi-os/blob/master/src/lesson01/src/boot.S](url)

rockytriton, 2021-02-03:
Ok yeah if that's the case, there isn't really any need to pre-define it for the assembly code to use it.  It just references a symbol by the given name and then the linker will try to find it.  For instance, if you rename the function kernel_main2 and then try to rebuild, it will fail during the linking process as the symbol won't be found.

xubenji, 2021-02-03:
> Ok yeah if that's the case, there isn't really any need to pre-define it for the assembly code to use it. It just references a symbol by the given name and then the linker will try to find it. For [...]

If I change the kernel_mian() to kernel_main2(). What I have to do to let the linker find the new function name?

xubenji, 2021-02-03:
> I think xubenji question might actually be about calling C functions from assembly files (not the other way around). Most of the time one would need to include the C header in the assembly file in order to call a function.
> 
> That said, I don't know how the boot.S file ([https://github.com/s-matyukevich/raspberry-pi-os/blob/master/src/lesson01/src/boot.S](url)) is able to call kernel_main without including any header that contains a reference to it.

Yes, It is what I thought

rockytriton, 2021-02-03:
> If I change the kernel_mian() to kernel_main2(). What I have to do to let the linker find the new function name?

The linker will know the name change because it sees the function name in the symbols list.  So as long as you change it in your .c file and your .S file then you will be good.

xubenji, 2021-02-03:
> The linker will know the name change because it sees the function name in the symbols list. So as long as you change it in your .c file and your .S file then you will be good.

It seems like I don't have to do too many operations. But, If the CPU wants to call some functions in .S files from .c files, we need to add the necessary header files, right? In this files, [https://github.com/s-matyukevich/raspberry-pi-os/blob/master/src/lesson01/src/mini_uart.c](url). I find it includes the function name which locals in utils.S