s-matyukevich/raspberry-pi-os #229: Resources to learn more about upcoming lessons

ginocontes, 2021-03-17:
Hi @s-matyukevich thanks for the first six lessons, they were really eye opening! I enjoyed every bit ( literally)  😄 
I don't know if you are still working on the next lessons but in the meantime can you suggest me some links or books to learn more about each of them?

rhett511, 2021-05-11:
I think you can research more useful resource at https://wiki.osdev.org/Main_Page

Kristine1975, 2021-11-12:
These tutorials are for Raspberry Pi 4, and one of the more recent ones is about networking: https://github.com/isometimes/rpi4-osdev

These are for Raspberry Pi 3, but they are mostly code, and very little explanations: https://github.com/bztsrc/raspi3-tutorial