s-matyukevich/raspberry-pi-os #237: [Lesson03] RPi 4 Interrupt not working

maxstreitberger, 2021-05-28:
I have some problems with converting the code for the RPi 3 to the RPi 4. I think I narrowed it down to the point where there is a problem with the handle_irq() function that doesn't get fired. But I don't know why it doesn't get called. I also tried to follow this [video](https://www.youtube.com/watch?v=nUW1FB_5vqo) which helped a little, but it still doesn't work.
Here are the functions I have in irq.c:
```
void enable_interrupt_controller() {
    put32(ENABLE_BASIC_IRQS, SYSTEM_TIMER_IRQ_1);
    printf("%x", get32(ENABLE_BASIC_IRQS));
}

void show_invalid_entry_message(int type, unsigned long esr, unsigned long address) {
    printf("%s, ESR: %x, address, %x\r\n", entry_error_messages[type], esr, address);
}

void handle_irq(void) {
    printf("Handling");
    unsigned int irq = get32(IRQ_BASIC_PENDING);
    switch (irq) {
    case (SYSTEM_TIMER_IRQ_1):
        handle_timer_irq();
        break;
    default:
        printf("Unknown pending irq: %x\r\n", irq);
        break;
    }
}
```

This is the irq.h file I use:
```
#define IRQ_BASIC_PENDING             (PBASE+0x0000B200)
#define IRQ_PENDING_1                       (PBASE+0x0000B204)
#define IRQ_PENDING_2                      (PBASE+0x0000B208)

#define ENABLE_BASIC_IRQS	         (PBASE+0x0000B210)
#define ENABLE_IRQS_1	                 (PBASE+0x0000B214)
#define ENABLE_IRQS_2	                 (PBASE+0x0000B218)

#define DISABLE_BASIC_IRQS	         (PBASE+0x0000B220)
#define DISABLE_IRQS_1	                 (PBASE+0x0000B224)
#define DISABLE_IRQS_2	                 (PBASE+0x0000B228)

#define SYSTEM_TIMER_IRQ_0	         (1 << 0)
#define SYSTEM_TIMER_IRQ_1	         (1 << 1)
#define SYSTEM_TIMER_IRQ_2	         (1 << 2)
#define SYSTEM_TIMER_IRQ_3	         (1 << 3)
```

maxstreitberger, 2021-07-08:
I found out that the raspberry pi 4 uses the **GIC-400 interrupt controller**. I used the [CoreLinkTM GIC-400 Generic Interrupt Controller Manual](https://developer.arm.com/documentation/ddi0471/b/), [ARM Generic Interrupt Controller Architecture Specification](https://developer.arm.com/documentation/ihi0069/d), and the [BCM2711 ARM Peripherals Manual](https://datasheets.raspberrypi.org/bcm2711/bcm2711-peripherals.pdf) to get information about it. 

To implement the timer interrupt we need the following information:
- The base address of the GIC-400
- The address for the Distributor Base
- The address for the CPU Base
- Enable the Interrupt Set-Enable Registers
  - Allow us to forward the corresponding interrupt to the CPU interfaces.
- Interrupt Acknowledge Register
  - Tell the processor which interrupt fired.
- End Of Interrupt Register
  - Indicates to the Distributor that it can forward the new highest priority pending interrupt
- Interrupt Processor Targets Registers
  - Tells the Distributor to which CPU interface it should forward the interrupt
- Time interrupt ID
  - VideoCore interrupts start at the interrupt ID of 96 (see page 87 in the BCM2711 ARM Peripherals Manual). If you look at the table on page 86 of the BCM2711 ARM Peripherals Manual, we need the index 0, 1, 2, and 3. Which then corespondents to the IDs of 96, 97, 98, and 99.

The irq.h file would then look like this:
```
#ifndef _P_IRQ_H
#define _P_IRQ_H

#include <peripherals/base.h>

#define GIC_BASE 0xFF840000
#define GICD_DIST_BASE (GIC_BASE+0x00001000)
#define GICC_CPU_BASE (GIC_BASE+0x00002000)

#define GICD_ENABLE_IRQ_BASE (GICD_DIST_BASE+0x00000100)

#define GICC_IAR (GICC_CPU_BASE+0x0000000C)
#define GICC_EOIR (GICC_CPU_BASE+0x00000010)

#define GIC_IRQ_TARGET_BASE (GICD_DIST_BASE+0x00000800)

//VC (=VideoCore) starts at 96
#define SYSTEM_TIMER_IRQ_0 (0x60) //96
#define SYSTEM_TIMER_IRQ_1 (0x61) //97
#define SYSTEM_TIMER_IRQ_2 (0x62) //98
#define SYSTEM_TIMER_IRQ_3 (0x63) //99

#endif
```

The irq.c looks then as follows:

```
#include "utils.h"
#include "printf.h"
#include "timer.h"
#include "entry.h"
#include "peripherals/irq.h"
#include "arm/sysregs.h"

const char *entry_error_messages[] = {
	"SYNC_INVALID_EL1t",
	"IRQ_INVALID_EL1t",		
	"FIQ_INVALID_EL1t",		
	"ERROR_INVALID_EL1T",		

	"SYNC_INVALID_EL1h",		
	"IRQ_INVALID_EL1h",		
	"FIQ_INVALID_EL1h",		
	"ERROR_INVALID_EL1h",		

	"SYNC_INVALID_EL0_64",		
	"IRQ_INVALID_EL0_64",		
	"FIQ_INVALID_EL0_64",		
	"ERROR_INVALID_EL0_64",	

	"SYNC_INVALID_EL0_32",		
	"IRQ_INVALID_EL0_32",		
	"FIQ_INVALID_EL0_32",		
	"ERROR_INVALID_EL0_32"	
};

void enable_interrupt(unsigned int irq) {
	 printf("%x\r\n", irq);
	 unsigned int n = irq / 32;
	 unsigned int offset = irq % 32;
	 unsigned int enableRegister = GICD_ENABLE_IRQ_BASE + (4*n);
	 printf("EnableRegister: %x\r\n", enableRegister);
	 put32(enableRegister, 1 << offset);
}

void assign_target(unsigned int irq, unsigned int cpu) {
	 unsigned int n = irq / 4;
	 unsigned int targetRegister = GIC_IRQ_TARGET_BASE + (4*n);
	 // Currently we only enter the target CPU 0
	 put32(targetRegister, get32(targetRegister) | (1 << 8));
}

void show_invalid_entry_message(int type, unsigned long esr, unsigned long address) {
	 printf("%s, ESR: %x, address, %x\r\n", entry_error_messages[type], esr, address);
}

void enable_interrupt_controller() {	
	 assign_target(SYSTEM_TIMER_IRQ_1, 0);
	 enable_interrupt(SYSTEM_TIMER_IRQ_1);
}

void handle_irq(void) {
	 unsigned int irq_ack_reg = get32(GICC_IAR);
	 unsigned int irq = irq_ack_reg & 0x2FF;
	 switch (irq) {
	 	 case (SYSTEM_TIMER_IRQ_1):
	 	 	 handle_timer_irq();
	 	 	 put32(GICC_EOIR, irq_ack_reg);
	 	 	 break;
	 	 default:
	 	 	 printf("Unknown pending irq: %x\r\n", irq);
	 	 	 break;
	}
}
```

rhythm16, 2021-07-31:
@maxstreitberger thanks for your effort, but do you know why doesn't Low Level Devel need to configure the GIC?
I (rpi4b) also followed the youtube video and everything works fine. By now I'm just confused. The bcm2711 peripheral datasheet doesn't explain clearly, 
what are the ARMC registers?
why do the ARMC registers control the VC interrupts when on page 87 the VC interrupts are connected to the GIC?
why is ARMC also listed as an interrupt source?
should I configure both the ARMC registers and the GIC?
so many questions... I'd be very thankful if anyone can help.

rhythm16, 2021-08-02:
hello past self, here's what I think is happening after some testing:
Low Level Devel doesn't need to configure the GIC because he used a custom `armstub` to start the RPI4, which does not enable the GIC. The default `armstub` probably sets up the GIC which causes the "ARMC" interrupt handling code to fail. An interesting point is that according to the [rpi docs](https://www.raspberrypi.org/documentation/configuration/config-txt/boot.md), the GIC is enabled by default, but once I remove the GIC enabling code in the `armstub`, the ARMC code works again, I guess this means that the GIC needs to be explicitly enabled in the `armstub`.
With all this, I think ARMC is a part of the legacy interrupt controller, but still I don't know what "ARMC" stands for.
A good reference for GIC enabling code can be found in [raspberrypi/tools](https://github.com/raspberrypi/tools/blob/master/armstubs/armstub8.S).

maxstreitberger, 2021-08-13:
@rhythm16 I hope I can provide some answers to your questions. 
1. I think ARMC stands for ARM-Core, which is used to send interrupts to **all cores** and you can find the registers in Table 117. ARMC Interrupt Registers in the [BCM2711 ARM Peripherals Manual](https://datasheets.raspberrypi.org/bcm2711/bcm2711-peripherals.pdf)
2. As for my understanding, the ARMC does not support VC interrupts (see Table 117. ARMC Interrupt Registers in the [BCM2711 ARM Peripherals Manual](https://datasheets.raspberrypi.org/bcm2711/bcm2711-peripherals.pdf) and page 83)
3. I think you don't have to configure both since all interrupts go through the GIC (if I interpret the diagram on page 87 in the  [BCM2711 ARM Peripherals Manual](https://datasheets.raspberrypi.org/bcm2711/bcm2711-peripherals.pdf) correctly)

Please correct me if I'm wrong somewhere.

closed 2021-09-06 by maxstreitberger

freewilll, 2023-08-02:
Brilliant. I spent hours trying to figure out how to get timer interrupts working on my raspberry pi 4b. The GIC-400 initialization and interrupt setup code here is what I needed. Thanks!