s-matyukevich/raspberry-pi-os #239: [Lesson05] RPi 4 system call not working

maxstreitberger, 2021-09-06:
I have some trouble getting system calls to work on my raspberry pi 4. When I run the OS I get the following error: 
\
`Kernel process started. EL 1 Type: 4, ESR: 2000000, Address: 1d14`
("Type: 4" means that it is "SYNC_INVALID_EL_1h") 
\
I looked at the [AArch64 Expection Model](https://developer.arm.com/documentation/102412/0100/Exception-types) to learn more about exceptions, but it didn't really helped me. I also tried the solution in the other issue (#228), which also didn't work for me.
Can anyone explain me what this error exactly means?

rhythm16, 2021-09-07:
Do the function pointers in your `sys_call_table` point to the correct place at runtime? I mean since `sys_call_table` is initialized statically, if you didn't change the linker script of modify `sys_call_table` at runtime, it would not be pointing to the `sys_*` functions because rpi4 loads the kernel image at `0x80000`

maxstreitberger, 2021-09-08:
Thanks for you quick response, I haven't changed the addresses of the `sys_*` functions. It is working now, after I changed it. Thank you!

closed 2021-09-08 by maxstreitberger

anarchuser, 2021-09-09:
This issue can also be resolved by loading the kernel code at address 0x80000 explicitly, by specifying
```
. = 0x80000;
```
as first line in `SECTIONS` in the `linker.ld`