s-matyukevich/raspberry-pi-os #252: why not used c in utils.S

merelyaspirant, 2022-06-03:
In lesson 1, any specific reason of not using C to implement put32 and get32 in utils.
Please clarify if its just a choice or any specific reason.

YBHQ, 2022-08-22:
~~~~

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

seusher, 2023-04-30:
For anyone who is wondering the same: utils.S is an assembly source file, so you can't directly use C code in it.  You can write the uart code all in C if you want to - see https://github.com/bztsrc/raspi3-tutorial/blob/master/03_uart1/uart.c

I'm not the author, but using assembly instead of C is a better educational experience and helps you understand ARM64 calling conventions rather than glossing over them and letting the compiler figure out the low-level details for you.