s-matyukevich/raspberry-pi-os #254: [exercises lesson06] after switch to user_process , it cannot response svc call

cuijier, 2022-06-09:
@xdfm1 Hi xdfm1, i haved seed your exercises  about lesson06, it's work , you are so clever. but according to your code, i have modified little , after run user_process , it can't response svc ，
```
(gdb) 
#0  0x0000000000085200 in ?? ()
#1  0x000000000000003c in user_process () at src/user.c:17
#2  0xffff0000000865f8 in ?? ()
```
can you help me to look it ?
the code link is [here](https://github.com/cuijier/learn_kernel/tree/v1)

YBHQ, 2022-08-22:
~~~~

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

YBHQ, 2022-08-22:
#~~~~

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

akshataj96, 2023-06-10:
If this is still open, can i start looking into it ? @cuijier